OctoPrint/octoprint-docker #213: [BUG] Container won´t start without USB device connected

wassereimer86, 2021-12-04:
**Describe the bug**
Container won´t start if the USB Device is not connected.

**Container Details**
2021-11-02T17:31:14.962Z

**To Reproduce**
1. Stop container
2. Diconnect USB device
3. Start container
4. Get the error

**Expected behavior**
Container start without USB device

**Error Message**
ERROR: for octoprint  Cannot restart container xxx: error gathering device information while adding custom device "/dev/ttyUSB0": no such file or directory

ghostsquad, 2022-05-07:
Can you provide full repro steps for this?

wassereimer86, 2022-05-10:
1. Start Container with assigned device /dev/ttyUSB0 (in Portainer or via command line) but not with any USB plugged in
3. Get the error

justindelliott, 2022-05-10:
This only happens on the container I'm using with my Prusa Mini - the Mini must first be powered on when the container starts up. Then I can turn off the Mini and it stays running. But any time the container is restarted I have to turn on the Mini again.

This does not happen with my Prusa Mk3 though. It never needs to be powered up for the container to start up.

Both of my containers are hosted on the same Linux server running Centos 7. All other configs for the environments are configured the same way, just updated paths to their respective USB /dev/ paths.

It could be related to the USB controller on the printer and how it handles USB when it's powered off?

wassereimer86, 2022-05-13:
> What command are you running to start the container? That's what I meant when I asked for full repro steps.

I use Portainer. So i can just say my settings.

- External Port 8124 to Internal 80
- Image: octoprint/octoprint:latest
- Volume octoprint to Container "/octoprint"
- Name "octoprint"
- Custom Network "docker"
- Env ENABLE_MJPG_STREAMER "true"
- Env MJPG_STREAMER_INPUT "-r 1920x1080 -f 30"
- Env CAMERA_DEV "/dev/video0"
- Restart always
- Device /dev/video0 to /dev/video0
- Device /dev/snd/controlC1 to /dev/snd/controlC1
- Device /dev/video1 to /dev/video1
- Device /dev/input/event8 to /dev/input/event8
- Device /dev/ttyUSB0 to /dev/ttyUSB0

Thats it.

I have an Anycubic i3 Mega. I have the same problem as @justindelliott - If i don´t have the printer powered on, the container won´t start. If i´ve powered it on, it starts and after that i can turn it off again.

ghostsquad, 2022-05-14:
I think this is an issue with docker. If you set to restart always, does
that work? Basically have docker try to restart the container continuously,
and when the printer turns on, then octoprint does too? Or are you saying
you want octoprint to be able to start whether the printer is on or not?

Cheers,
Wes

Paraphraser, 2022-05-14:
I don't know if anything in [this gist](https://gist.github.com/Paraphraser/650eeab26794dc7363b7123b33f671c4) will prove useful. I've had mine set up that way for over a year.

wassereimer86, 2022-05-14:
> Or are you saying you want octoprint to be able to start whether the printer is on or not?

This. I want to configure and upload things to octoprint when the printer is turned off. If the container isn´t running and the printer is off, i have to turn it on to start the octoprint container. Otherwise it won´t start.

The container won´t even log something which i can present to you.

tvervest, 2023-02-21:
Hey all, I faced this exact issue and figured I should share my conclusion/solution.

As it turns out, the problem is that docker containers can't have a device mounted at runtime, period. At the same time, as the USB device simply doesn't exist yet when the printer isn't connected, it's impossible to bind the device to the container at boot time. This is what causes the container to fail to boot up. This is not an issue with the image, it is a natural result of how binding works in docker.

The solution is to mount the `/dev` folder to the docker container in stead. As a result of doing this, the docker container will be able to boot (as the mounted folder exists at boot time) and new devices are accessible to the container, as it has full access to the `/dev` folder and all of it's content.

Wether it's a good idea or not to give a container that's running an image you don't own is of course the logical next question, but you'll have to make that decision for yourself. In my case I decided I could live with the risk 😉

closed 2023-02-26 by LongLiveCHIEF