OctoPrint/octoprint-docker #214: octoprint wont start:

jonwilliams84, 2021-12-21:
I am trying to run this image in a kubernetes cluster using virtualhere client on the node that I am pinning the pod to in order to mount the USB device (printer) connected remotely to another pi-zero.

I have tested this and it worked perfectly for the first time the pod runs. 

Here is my deployment.yaml:
```
apiVersion: apps/v1
kind: Deployment
metadata:
  creationTimestamp: null
  labels:
    app: octoprint
  name: octoprint
  namespace: octoprint
spec:
  replicas: 1
  selector:
    matchLabels:
      app: octoprint
  strategy: {}
  template:
    metadata:
      creationTimestamp: null
      labels:
        app: octoprint
    spec:
      nodeName: tcl-k8s-1
      containers:
      - image: octoprint/octoprint
        name: octoprint
        securityContext:
          privileged: true
        volumeMounts:
          - mountPath: /octoprint
            name: datastore
          - mountPath: /dev/ttyACM0
            name: ttyacm
      volumes:
       - name: ttyacm
         hostPath:
           path: /dev/ttyACM0
       - name: datastore
         persistentVolumeClaim:
           claimName: pvc-octoprint
```
The pvc (using longhorn):
```
apiVersion: v1
kind: PersistentVolumeClaim
metadata:
  name: pvc-octoprint
  namespace: octoprint
spec:
  accessModes:
    - ReadWriteOnce
  storageClassName: longhorn
  resources:
    requests:
      storage: 1Gi
```

As mentioned the first time this spun up it worked flawlessly. The node where the pod is scheduled has the printer connected via VirtualHere:
```
VirtualHere Client IPC, below are the available devices:
(Value in brackets = address, * = Auto-Use)

Raspberry Hub (octopi:7575)
  *--> Maple (octopi.115) (In-use by you)
```
lsusb:
```
Bus 009 Device 008: ID 1eaf:0029 LeafLabs Maple
```
...and Octoprint worked and picked up the /dev/ttyACM device and I was able to do a test print without any issue, as if the host was directly connected to the printer.

Fast forward a couple of days, now when the pod comes up, I try and hit the octoprint service that I have exposed and added ingress for which was working perfectly, gets a 503 error page. 

The logs of the pod are just being spammed with:
```
Error: Invalid value for '--port': tcp://10.43.71.129:80 is not a valid integer
Usage: octoprint [OPTIONS] COMMAND [ARGS]...
Try 'octoprint --help' for help.
```

So, at first I think, maybe the PVC/data held in the volume is corrupt somehow. I delete the deployment, and the pvc and then recreate it. Same error. No matter what I try, the octoprint service within the container never comes up.

I have checked the s6 service supervisor / controller and the command its using to launch octoprint looks fine:
```
root@octoprint-66fbd6d67b-mgft2:/# cat /var/run/s6/services/octoprint/run
#!/usr/bin/with-contenv sh

exec octoprint serve --iknowwhatimdoing --host 0.0.0.0 --port 5000 --basedir /octoprint/octoprint
```

Any thoughts?  I know this is probably not the usual deployment method!

stepanovdg, 2022-01-06:
is the same to https://github.com/OctoPrint/octoprint-docker/issues/214

Rename service from octoprint  (works for me) and change service port from 80 to 5000

jonwilliams84, 2022-01-12:
> is the same to #214
> 
> Rename service from octoprint (works for me) and change service port from 80 to 5000

I have renamed the service, I assume you mean the service listener port not the target?

jonwilliams84, 2022-01-12:
Ok, you are indeed correct, changing the service name and updating the service listener port does indeed resolve the issue.

now I need to work out why octoprint can't connect to the ttyACM0 device I have shared over VirtualHere, that was working perfectly previously!

stepanovdg, 2022-01-15:
@jonwilliams84 still the issue sorry.
I had same problems as you so first was changing service port and that worked for octoprint itself.
After buying camera I got problems with mijpg-streamer on port 8080
with modifying values-override and trying to work with common chart library rules
finished with creating second service for port 8080
and manually creating ingress for that service with path /webcam - as common chart ignores config and always map all backend to main service.

After that was able to see webcam stream from https://mydomain/webcam/?action=strream
And was able to see screenshots for timelapse in octoprint itself. But trying to use that url for octoprint stream - just was making ui dying and never loading. Backend service working fine. 

So after I took a look inside pod itself to restart service and remove stream manually from file I noticed
that container has HA proxy running on port 80 for /webcam mapping to 8080 and all other to port 5000.

So had to remove all my service changes for port 5000 
and leave it as is for port 80 

In this configuration -> octoprint working , screenshot working, but video stream is not working 
just black window with
Currently configured stream URL: /webcam?action=stream

If you can succeed with configuring this please comment )

wlritchi, 2022-03-13:
I believe this is a duplicate of #160 (which has a workaround)

wlritchi, 2022-03-13:
Actually, on further investigation, it looks like the fix for #160 is incomplete. Here's a (trimmed) sample of a few attempts at running the latest Octoprint (installed via pip, non-Docker):

```
$ octoprint serve --port 22222
...
2022-03-13 01:08:01,529 - octoprint.server - INFO - Listening on http://0.0.0.0:22222 and http://[::]:22222
...

$ OCTOPRINT_PORT=12345 octoprint serve
...
2022-03-13 01:08:29,735 - octoprint.server - INFO - Listening on http://0.0.0.0:12345 and http://[::]:12345
...

$ OCTOPRINT_PORT=12345 octoprint serve --port 22222
...
2022-03-13 01:08:37,920 - octoprint.server - INFO - Listening on http://0.0.0.0:22222 and http://[::]:22222
...

$ OCTOPRINT_PORT=eeee octoprint serve --port 22222
Usage: octoprint [OPTIONS] COMMAND [ARGS]...
Try 'octoprint --help' for help.

Error: Invalid value for '--port': eeee is not a valid integer
```
To summarize, the CLI parameter takes precedence over the env var (as it should), but if the env var isn't a valid integer, OctoPrint will fail to launch *even if the parameter is provided and should take precedence*. As far as I can tell the env var handling is all coming directly from [Click](https://click.palletsprojects.com/en/8.0.x/). I may do some more testing and report it upstream, but in the meantime, we might want to consider one of the other fixes/workarounds proposed in the discussion of #160.