OctoPrint/octoprint-docker #224: Q: How do I get the webcam working usign ngrok?

jorgegomzar, 2022-05-08:
## Question/Problem Description

I moved from Octopi to this octoprint container. Everything works fine except for the fact that I cannot get the webcam to work through a ngrok tunnel (using the octoprint plugin).
May it be related to a firewall inside the octoprint image? Or may it be related to the docker config of the network used?

### System Details

I'm running octoprint docker using the `docker-compose.yml` file:
[image]

[image]

franz134, 2022-05-12:
I have exactly the same issue.
First I read out the devices as shown and strangely got 2 video devices, despite only one webcam is plugged in. (also tried a usb-capturecard - also 2 video devices recognized, same issue)
[image]
In the .yaml file I use "video0" but I also tried video1 - no difference.
[image]
So I got octoprint up and running, but no webstream in the control tab.
[image]
However I have made an interesting discovery: the feature of making timelapse is working and in the settings I can test the snapshots, but I have no clue why the stream isn't working.
[image]

franz134, 2022-05-16:
I've found the solution, at least for my problem, after quite a while of research in the octoprint forum:
https://community.octoprint.org/t/octoprint-in-docker-snapshot-works-stream-doesnt/43699
I had to write `ports: - 5000:80` instead of my `5000:5000`
Not sure if this also fixes your issue, but maybe it's worth trying.

closed 2022-05-20 by LongLiveCHIEF