OctoPrint/octoprint-docker #226: Support to encode timelapse videos with H.265 / HEVC

epheterson, 2022-06-10:
**Describe the solution you'd like**
To have smaller timelapse video files to share when the print is done, it'd be great if we could start encoding with H.265 / HEVC. This will result in essentially the same quality video, but much smaller file size.

I understand the ffmpeg command line is now editable, but don't believe the Octoprint docker package ships with ffmpeg capable of h.265 encoding, so it would have to be re-compiled. Ideally H.265 could be a simple selection in the drop-down of advanced timelapse settings.

**Additional context**
Asked on OctoPrint github and was directed to send the request here:
https://github.com/OctoPrint/OctoPrint/issues/4538

LongLiveCHIEF, 2022-06-15:
Actually, i missed the part where it said `x265` needs to be installed on the system, and that's easy enough to fix by adding that package to the `apt install` step of the `Dockerfile`.  I can definitely add that.

LongLiveCHIEF, 2022-06-15:
I've merged and deployed a new image with `x265` dependency in the image. That should be all that was mising. Once the new image finishes building and deploying, can you pull it down and give this a shot?

Thanks!

epheterson, 2022-06-16:
Thanks, giving it a shot. I'm not positive what the best command is to use, some considerations:
- Setting a constant bitrate will result in not necessarily saving any on the filesize. (To workaround this, I removed bitrate altogether to utilize default crf.)
- Built-in variables for container format and filters may not be compatible
- I can only test once per print

Do you know of a way to re-run the timelapse rendering so I can test multiple settings? Also do you know how to add the codec to the drop-down list for easy user selection? (I'd prefer to not have to use a custom ffmpeg command)

Starting with:
`{ffmpeg} -framerate {fps} -i "{input}" -vcodec libx265 -threads {threads} -f {containerformat} -y {filters} "{output}"`

Think I might want to try something closer to what [the docs](https://trac.ffmpeg.org/wiki/Encode/H.265) suggest, though:
`ffmpeg -i input -c:v libx265 -crf 26 -preset fast -c:a aac -b:a 128k output.mp4`

epheterson, 2022-06-16:
Well, my first one made a file, and the file size is ~45% of MP4 size, but it doesn't play in Quicktime, though does play in VLC.

`ffmpeg -i` shows it's HEVC, correct framerate/resolution and 4092 kb/s (vs. default of 10000 kb/s):
`Stream #0:0[0x1](und): Video: hevc (Main) (hev1 / 0x31766568), yuv420p(tv, progressive), 1920x1080, 4092 kb/s, 25 fps, 25 tbr, 12800 tbn (default)`

Going to see if I can figure out why Quicktime / browser won't play it.

epheterson, 2022-06-16:
Got it! For hevc this must be included: `-vtag hvc1`

So the full modified line, with bitrate removed and `-vtag hvc1` added is:
`{ffmpeg} -framerate {fps} -i "{input}" -vcodec libx265 -vtag hvc1 -threads {threads} -f {containerformat} -y {filters} "{output}"`

Thanks for adding the codec support! Do you know how we can go about making this an "official" option that's easy to select?

LongLiveCHIEF, 2022-06-20:
Well, if you're interested in doing a PR, then here's what I'd probably do.

- create an env var for `CAMERA_H265`
- modify the [mjpg-streamer](https://github.com/OctoPrint/octoprint-docker/blob/master/root/etc/services.d/mjpg-streamer/run) service script so it tests for this variable, and if set, then uses those args to start instead of the default

This is the basic approach mind you, there might be more logic needed there.

If this isn't something you can contribute, go ahead and open a feature request issue for `add CAMERA_H265 env var start option`, and link to this video, and I'll try to get to it.

epheterson, 2022-06-20:
Hey @LongLiveCHIEF thanks for the steps though I'm totally sure how to go about it. I've filed a new issue as you suggested: https://github.com/OctoPrint/octoprint-docker/issues/228