OctoPrint/octoprint-docker #228: Add CAMERA_H265 env var start option

epheterson, 2022-06-20:
Follow-up for work added in:
https://github.com/OctoPrint/octoprint-docker/issues/226

Please add the H265 encoder as a default option to the drop-down encoder list in Octoprint. This requires both changing the encoder codec to `libx265` and adding an additional variable `-vtag hvc1`:
`-vcodec libx265 -vtag hvc1`

LongLiveCHIEF, 2022-06-26:
Changing UI elements is not something that can be done in the docker image.

closed 2022-06-26 by LongLiveCHIEF

epheterson, 2022-06-28:
@LongLiveCHIEF You requested I file this ticket, does it belong in the Octoprint repo instead?

LongLiveCHIEF, 2022-07-04:
My suggestion was to open a request to modify the docker container env vars and service scripts. This request you opened was about changing the UI.

This is the docker project, and deals with the docker image. That means no changes to OctoPrint source code.