OctoPrint/octoprint-docker #229: [BUG] docs/README.md: broken link to docs/preconfigure-your-octoprint.md

jmtd, 2022-06-24:
The page [docs](https://github.com/OctoPrint/octoprint-docker/tree/master/docs)/README.md contains a link to docs/preconfigure-your-octoprint.md which does not exist.

Artmorse, 2022-08-22:
Actually, the documentation has been merged with the [docs/README.md file](https://github.com/OctoPrint/octoprint-docker/tree/master/docs#preconfiguring-your-octoprint-container) (in the *Preconfigure Your OctoPrint Container* section). 
You can retrieve the file in the older commits: [here](https://github.com/OctoPrint/octoprint-docker/commit/f86332d47b1e0eddca533ba6d470f1ff0a26c6ec).

I think the link needs to be removed.

Artmorse, 2022-08-22:
I've opened a Pull Request (#231).

closed 2022-09-25 by LongLiveCHIEF