OctoPrint/octoprint-docker #232: Compatibility with old webcam

dormancygrace, 2022-09-02:
**Is your feature request related to a problem? Please describe.**
I have old camera ID 093a:2620 Pixart Imaging, Inc. It works under Ubuntu 20.04, but only when i use *LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so fswebcam* for example.

**Describe the solution you'd like**
checkbox or env setting for use LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so for camera

aguinaldoabbj, 2022-09-22:
As your problem is very specific, isn't it better to just create a new Dockerfile based on this current Dockerfile with this env variable set?

LongLiveCHIEF, 2023-02-26:
We can't support every camera out there, as we have no way to test. As @aguinaldoabbj suggested, I'd recommend creating your own Dockerfile using octoprint/octoprint as a base, and add that lib and env var setting.

closed 2023-02-26 by LongLiveCHIEF