OctoPrint/octoprint-docker #233: Container working in IPv6 and IPv4 network envoirements

MaZe3D, 2022-09-11:
**Is your feature request related to a problem? Please describe.**
The docker only listens on a IPv4 address. It was tried to solve with #175 but broke the configuration of others #192.
**Describe the solution you'd like**
The docker should be capable to be used in a IPv4/IPv6 only or Dual-Stack envoirement.
**Describe alternatives you've considered**
I had a look at the #175 and #192. I manually changed it to the original pull request.
**Additional context**
I have a IPv6 only network at home and want to use the printer in this network. I belief that this container should work for everyone. I dont want you to change your netowork as much as I dont want to hear from you that I should change my network configuration. (Tis should not be a v4/v6-war as this discussions tent to get there).

Thanks everyone for reeding my issue. I wish everyone a lovely day.

LongLiveCHIEF, 2023-02-26:
The whole point of docker is so that image maintainers can make an image work inside a container however they want in regards to networking and such, and operations (the person deploying the _container_) can configure their networks however _they_ want.

There's nothing that can be done inside this image that can determine whether or not the _container_ listens on IPv4 or IPv6. Unfortunately, that _is_ a network change on your part to map IPv6 network you're binding the container to to map to the resulting port. It doesn't matter if the container is using IPv4 inside itself, you can still use IPv6 to map to the container if you configure your host to do so.

MaZe3D, 2023-02-27:
This is only true if the Docker is used in Bridge Mode. If the docker has assigned a IP Address from within a Network (not dockers own) you actually assign a IPv4 or IPv6 and the container needs to be able to handle them. 
In this case if I assign a static IPv6 to the container, I cannot access the Web GUI. And that is a real bummer if you are working in a IPv6 only network.

LongLiveCHIEF, 2023-02-27:
I may have more to learn on this subject. 

@MaZe3D would you be able to do a test for me? In one of the PR's that you mentioned, we had attempted to implement IPv6 support for the nginx config, but it was causing some issues for users right away. 

Can you create a container of your own, and change the line https://github.com/OctoPrint/octoprint-docker/blob/master/root/etc/haproxy/haproxy.cfg#L23 to be

```cfg
        bind *:80 v4v6
```

and see if that works for you? If so, we'll go from there. 

A few questions just to help me:

- do you have any other containers running on this IPv6 only system that you access over TCP/IP?
- do you have the [docker daemon configured to support IPv6](https://docs.docker.com/config/daemon/ipv6/)?

MaZe3D, 2023-03-05:
This is an issue that also other containers encounter. I investigated and had no perfect solution. But recently, because of an similar issue form https://github.com/dani-garcia/vaultwarden/discussions/3306, I have now an Idea that should satisfy most setups.

What about adding a optional container Variable that will set the line like above. so you could self define the bind-address (and maybe even port) of the container.
Maybe like
```
BIND_ADDRESS=::
BIND_PORT=80
```

closed 2023-03-05 by LongLiveCHIEF

MaZe3D, 2023-03-05:
Is this now fixed, or why is it closed?

LongLiveCHIEF, 2023-03-05:
[Containers have no information about what kind of network](https://docs.docker.com/config/containers/container-networking) it's attached. It doesn't matter if it's an IPv6 or IPv4 network.  If you want to map an IPv6 network to a container, you can use port forwarding during container creation to forward the IPv6 host and port on whatever network your container is attached to.

For example, in your case:  `-p [::]:80:80` in your `docker run command, or:

```yaml
ports:
  - "[::]:80:80
```

In you docker-compose.yml should do the job (assuming you have IPv6 enabled on your network of choice and the [docker daemon](https://docs.docker.com/config/daemon/ipv6/).)

If you configure your host to support IPv6 (since your host is already an IPv6 only machine), then you won't have issues trying to get individual support for IPv6 for individual images.

Another great thing about docker, is that you can always use an official image as a base image, and change it as needed in your own Dockerfile, if you don't want to configure your host properly for IPv6 docker networks.

LongLiveCHIEF, 2023-03-05:
https://dev.to/joeneville_/build-an-ipv6-network-with-docker-compose-434i