OctoPrint/octoprint-docker #235: [BUG] Login fails after update to octoprint 1.8.3

Sroose, 2022-09-22:
**Describe the bug**
After letting octoprint update itself to the latest version, 1.8.3, logins fail (http 403)

It seems to be related to haproxy an security changes dicussed at https://github.com/OctoPrint/OctoPrint/issues/4645

**Container Details**
"2022-08-09T12:33:57.320Z"

**To Reproduce**
1. login to to octoprint web gui
2. allow updating to 1.8.3
3. try to login again after it restarts

DevAlphaKilo, 2022-10-06:
This issue is also in image `1.8.4`  and I'm having the same problem. Any recommendations for fixes?

drewgates, 2022-10-23:
Same problem. I've been poking at this for about a week now with no success.

darsys, 2022-12-17:
I had the same issue.  I am using docker-compose and the fix was to docker-compose down and up.  I know this sounds too simple but it worked for me.  I had restarted the container via docker several times.

LongLiveCHIEF, 2023-02-26:
One thing to note is that when using octoprint cli in this image, you have to pass the `--basedir /octoprint/octoprint` option so it looks in the correct directory for octoprint configurations and data.