OctoPrint/octoprint-docker #240: Config_Directory not properly initiated

hiramknick, 2022-10-12:
Octodash Companion fails when added to a Docker Instance.  

The following statement is in the configuration for OctoDash Companion:
`config_directory=normalize(_default_configdir("octodash")) `

Instead of routing to the default /octoprint referenced in the standard docker image:
```
    volumes:
     - octoprint:/octoprint
```
Instead, the extension is finding the following "/root/.config/octodash/config.json"

hiramknick, 2022-10-13:
See https://github.com/jneilliii/OctoPrint-OctoDashCompanion/issues/34#issuecomment-1276106192 for details

jneilliii, 2022-10-13:
This is something I can handle from the plugin side and don't think I need to involve @LongLiveCHIEF unless he has a good hint on how to detect if OctoPrint is running inside docker or not?

LongLiveCHIEF, 2022-10-13:
Well, for the normal container, you'd have s6 running, and that would only be possible inside the official Octoprint image, but I'm going to say we close this issue in favor of the issue above or another separate issue in the OctoDash repo, and I can collaborate with you further there or on discord. I could also do something easy like set a `OCTOPRINT_DOCKER=true` env var in each of the containers so you and other plugin authors have an easy way to detect if octoprint is running in an official container.

closed 2022-10-13 by LongLiveCHIEF