OctoPrint/octoprint-docker #242: [BUG] octoprint/octoprint:latest or :1.8.6 not up to date

fischer-felix, 2022-10-30:
**Describe the bug**
When deploying octoprint using the official image and the tag `latest` or `1.8.6`, octoprint itself shows the version as being 1.8.1.

**Container Details**

please run `docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:<tag_you_are_using>' and list the date returned.
2022-10-18T09:25:10.308Z

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy OctoPrint using `octoprint/octoprint:latest` or `octoprint/octoprint:1.8.6`
2. Open the webinterface
3. Go to Settings > About OctoPrint
4. See that Version is 1.8.1

**Expected behavior**
OctoPrint is _actually_ on version 1.8.6 as the tag suggests.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version: 106.0.2

Paraphraser, 2022-10-30:
I don't know if this will help but I'm not seeing the same pattern.

Same result from the "inspect":

```
$ docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:latest 
2022-10-18T09:25:10.308Z
```

When I search the image for a version:

```
$ docker image inspect octoprint/octoprint:latest | grep "1\.8"
                "octoprint_ref=1.8.6",
```

When I ask the process to tell me its version:

```
$ docker exec octoprint octoprint --version
octoprint, version 1.8.6
```

And when I look in the UI:

[image]

[image]

Paraphraser, 2022-10-31:
Hmmm. My compose file has

```
- ./volumes/octoprint:/octoprint
```

so that's probably close to what you are doing, right?

The "rules" I follow (I'm not sure whether they are actual rules or just how I approach it) are that, (1) if I get a prompt to update Octoprint, I wait for a new image to appear on DockerHub; (2) if I get a prompt to update a plug-in, I do that in the plug-in manager (which is persisted).

I learned that first rule the hard way and I'm pretty sure I had to nuke my persistent store and start over to recover.

fischer-felix, 2022-11-01:
Guess I'll see when the next update comes around

closed 2023-02-26 by LongLiveCHIEF