OctoPrint/octoprint-docker #243: Q: octoprint-docker on Windows

DasLukas, 2022-11-03:
## Question/Problem Description

I plan to run octoprint-docker on a Windows machine. I didn't find any descriptions for this use case. Before wasting time with testing I want to ask if it's possible to run this docker image on a windows machine.

In General I'm not safe with working on Windows as with Unix based systems. So has anyone experience to pass the right printer and webcam port to this docker image?

## Describe your hardware

Windows 10, Intel i7 Gen.

lin-ycv, 2022-11-16:
probably not, there' no usb passthrough in either WSL or hyper-v

LongLiveCHIEF, 2023-02-26:
This _might_ be possible by installing usbipd-win in wsl2 on Windows 11, which Microsoft details [docs and prerequisites for here](https://learn.microsoft.com/en-us/windows/wsl/connect-usb), but this isn't something we'd officially support, and there's no telling what device names would be, or if they would even work.

closed 2023-02-26 by LongLiveCHIEF