OctoPrint/octoprint-docker #249: Multiple processes of mjpg streamer started

eRaz3r, 2023-02-28:
Don't know since when but when I start my container of octoprint there are multiple processes for the mjpeg streamer. I absolutely don't know what this about but it seems not normal.

[image]
[image]

LongLiveCHIEF, 2023-03-05:
When was the last time you pulled the octoprint container?  I'm wondering if this was a bug introduced in https://github.com/OctoPrint/octoprint-docker/pull/245

eRaz3r, 2023-03-05:
> When was the last time you pulled the octoprint container? I'm wondering if this was a bug introduced in #245

There were two updates since I noticed this problem.

LongLiveCHIEF, 2023-03-05:
When was the last time you pulled the octoprint container?

LongLiveCHIEF, 2023-03-05:
I'm pretty sure this was introduce in #245, and that you are running an octoprint image that was pulled after that was pushed to docker hub.  I'm going to revert that change here shortly and publish the fix up to docker hub.

closed 2023-03-05 by LongLiveCHIEF

eRaz3r, 2023-03-05:
I hope you're right. I am running an octoprint docker on another server without this problem. I will pull the container and will see. Thank you very much for your fast reply.

eRaz3r, 2023-03-05:
> I'm pretty sure this was introduce in #245, and that you are running an octoprint image that was pulled after that was pushed to docker hub. I'm going to revert that change here shortly and publish the fix up to docker hub.

Problem seems to be fixed. Thank you very much. Any chance that a noob like me understands what the problem was about?

jniclas, 2023-03-08:
Hey, thank you for fixing this so fast! 👍🏻 
After pulling the image again, everything worked great again.
Because I was googleing a lot until I found this, because the logs are posted only as image, I shortly post my logs as text, so others may find this issue faster:

```
Starting mjpg_streamer for camera device /dev/video0 on port 8080
enabling daemon modeforked to background (4474)
Starting mjpg_streamer for camera device /dev/video0 on port 8080
enabling daemon modeforked to background (4481)
Starting mjpg_streamer for camera device /dev/video0 on port 8080
enabling daemon modeforked to background (4488)
Starting mjpg_streamer for camera device /dev/video0 on port 8080
enabling daemon modeforked to background (4495)
Starting mjpg_streamer for camera device /dev/video0 on port 8080
enabling daemon modeforked to background (4502)
...
```

PR #250 (2023-03-05): revert #245
Reverts #245 

fixes #249