OctoPrint/octoprint-docker #251: [BUG] Webassets are not loaded when X-Script-Name is set

wolkenschieber, 2023-03-12:
**Describe the bug**

Webassets are not loaed, when a custom path through `X-Script-Name` is set.

**Container Details**

My compose file looks like:
```yaml
version: '3.9'

services:
  octoprint:
    labels:
      - "traefik.enable=true"
      - "traefik.docker.network=proxy"
      - "traefik.http.routers.octoprint.entrypoints=web"
      - "traefik.http.routers.octoprint.rule=PathPrefix(`/octoprint`)"
      - "traefik.http.services.octoprint.loadbalancer.server.port=80"
      - "traefik.http.middlewares.opPathHeader.headers.customrequestheaders.X-Script-Name=/octoprint"
      - "traefik.http.routers.octoprint.middlewares=opPathHeader@docker"
    image: octoprint/octoprint
    restart: unless-stopped
    volumes:
     - ./config:/octoprint
    networks:
     - proxy

networks:
  proxy:
    external: true
```

**To Reproduce**

Navigate to http://SERVER/octoprint/. The page won't load. File from `static/webassets` report 404.

**Screenshots**

[image]

shotor, 2023-11-04:
I had the same issue. Came here to ask for more attention to this issue. But then I noticed a typo in my config.

In any case, to fix this issue you also need to strip the `/octoprint` prefix and add the `X-Scheme` header if you're using https. I'll add this to the community FAQ as well.

```yml
labels:
  - "traefik.http.middlewares.octoprint-headers.headers.customrequestheaders.X-Script-Name=/octoprint"
  - "traefik.http.middlewares.octoprint-headers.headers.customrequestheaders.X-Scheme=https"
  - "traefik.http.middlewares.octoprint-strip-prefix.stripprefix.prefixes=/octoprint"
  - "traefik.http.routers.octoprint.middlewares=octoprint-headers,octoprint-strip-prefix"
```

Lan-Hekary, 2024-03-27:
I have the same bug when I try to use it with the Path Prefix, is there a solution for it yet?

shotor, 2024-05-23:
> I have the same bug when I try to use it with the Path Prefix, is there a solution for it yet?

Try:

```
labels:
  - "traefik.http.middlewares.octoprint-headers.headers.customrequestheaders.X-Script-Name=/octoprint"
  - "traefik.http.middlewares.octoprint-headers.headers.customrequestheaders.X-Scheme=https"
  - "traefik.http.middlewares.octoprint-strip-prefix.stripprefix.prefixes=/octoprint"
  - "traefik.http.routers.octoprint.middlewares=octoprint-headers,octoprint-strip-prefix"
```