OctoPrint/octoprint-docker #252: [BUG] Loading error: Passive login failed.  raspi 4

hemna, 2023-03-22:
**Describe the bug**
Can't login or get normal first start web UI on octoprint start.

**Container Details**

please run `docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:<tag_you_are_using>' and list the date returned.

Container details
```
waboring@octoprint-pi4:~ $ docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint
2023-03-05T15:08:11.823Z
```

```
docker logs -f 3def9bbcbe83
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-mjpg-streamer-config: executing...
[cont-init.d] 01-mjpg-streamer-config: exited 0.
[cont-init.d] v3-dir-structure-check: executing...
[cont-init.d] v3-dir-structure-check: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
2023-03-22 18:53:53,765 - octoprint.startup - INFO - ******************************************************************************
2023-03-22 18:53:53,771 - octoprint.startup - INFO - Starting OctoPrint 1.8.7
2023-03-22 18:53:53,774 - octoprint.startup - INFO - ******************************************************************************
2023-03-22 18:53:54,400 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2023-03-22 18:53:54,458 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2023-03-22 18:53:54,458 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2023-03-22 18:53:54,458 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2023-03-22 18:53:54,477 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
2023-03-22 18:53:58,013 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 38 hook handlers
2023-03-22 18:53:58,149 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2023-03-22 18:53:58,305 - octoprint.server - INFO - Intermediary server started
2023-03-22 18:53:58,328 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /octoprint/octoprint/uploads...
2023-03-22 18:53:58,331 - octoprint.filemanager.storage - INFO - ... file metadata for /octoprint/octoprint/uploads initialized successfully.
2023-03-22 18:53:58,336 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2023-03-22 18:53:58,337 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2023-03-22 18:53:58,337 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2023-03-22 18:53:58,337 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2023-03-22 18:53:58,338 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2023-03-22 18:53:58,338 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2023-03-22 18:53:58,339 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
2023-03-22 18:53:58,339 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2023-03-22 18:53:58,340 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2023-03-22 18:53:58,340 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2023-03-22 18:53:58,340 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2023-03-22 18:53:58,341 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_LIST (needs: "Need(method='role', value='plugin_pluginmanager_list')")
2023-03-22 18:53:58,341 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2023-03-22 18:53:58,341 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2023-03-22 18:53:58,344 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2023-03-22 18:53:58,351 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2023-03-22 18:53:58,351 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2023-03-22 18:53:58,406 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2023-03-22 18:54:12,132 - octoprint.util.pip - INFO - Using "/usr/local/bin/python -m pip" as command to invoke pip
2023-03-22 18:54:13,385 - octoprint.util.pip - INFO - Version of pip is 22.0.4
2023-03-22 18:54:13,398 - octoprint.util.pip - INFO - pip installs to /usr/local/lib/python3.8/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no
2023-03-22 18:54:13,398 - octoprint.util.pip - INFO - ==> pip ok -> yes
[... 146 lines cut ...]
  ####
  # uncomment if you wish to edit the configuration files of octoprint
  # refer to docs on configuration editing for more information
  ####

  config-editor:
    image: linuxserver/code-server
    ports:
      - 8443:8443
    depends_on:
      - octoprint
    restart: unless-stopped
    environment:
      - PUID=0
      - GUID=0
      - TZ=America/New_York
    volumes:
      - ./octoprint:/octoprint

volumes:
  octoprint:
```

**Additional context**
Add any other context about the problem here.

hemna, 2023-03-24:
It looks like this is a problem with haproxy inside the container.
I opened up port 5000:5000 and hit that with my browser and the octoprint UI is working fine.

Afraskai, 2023-03-24:
I can confirm this, with port 5000 it works. Port 80 does not let you log in. I checked it also with a fresh blank container. The only thing which changed was the kernel on the Raspi4 with an update a few days ago:

uname -m via console showes it´s now:
aarch64

I tried roll it back in /boot/config.txt but this produced an instable OS so it´s not reproduce the problem.

hemna, 2023-03-24:
```
waboring@octoprint-pi4:~ $ uname -a
Linux octoprint-pi4 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```

szimonek, 2023-03-27:
I have the same problem. It appeared after dist-upgrade Raspbian from buster to bullseye. The workaround with exposing port 5000 and bypassing HA PROXY works.

Tracking down this problem I noticed that haproxy respond with connection reset (NS_ERROR_NET_RESET) to POST requests where GET requests are working fine. The first request that is failing is POST http://octopi/api/login (octopi is a DNS name of my raspberry pi)

I'm using latest stable image (1.8.7) 
My kernel and docker version are:
```
# uname -a
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
#
# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
#
# docker vesrion
docker: 'vesrion' is not a docker command.
See 'docker --help'
root@raspberrypi:~/octoprint-docker# docker version
Client:
 Version:           20.10.5+dfsg1
 API version:       1.41
 Go version:        go1.15.15
 Git commit:        55c4c88
 Built:             Mon May 30 18:34:49 2022
 OS/Arch:           linux/arm
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.5+dfsg1
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.15.15
  Git commit:       363e9a8
  Built:            Mon May 30 18:34:49 2022
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.4.13~ds1
  GitCommit:        1.4.13~ds1-1~deb11u3
 runc:
  Version:          1.0.0~rc93+ds1
  GitCommit:        1.0.0~rc93+ds1-5+deb11u2
 docker-init:
  Version:          0.19.0
  GitCommit:
#
# docker-compose version
docker-compose version 1.25.0, build unknown
docker-py version: 4.1.0
CPython version: 3.9.2
OpenSSL version: OpenSSL 1.1.1n  15 Mar 2022
#
```

possebaer, 2023-04-07:
I had the same issue and I used the workaround so far. 

Then I noticed PrintTimeGenius plugin stopped working as well. Analysing the problem, I saw that it calls an aarch64 binary, because "uname -a" reports "aarch64". The "marlin-calc.aarch64" binary did not work inside the docker container, but outside, on the raspi host it worked. As the PTG plugin had worked before, and stopped working at the same time I got the "passive login problem" I knew that the problem was introduced by a change from a 32bit kernel to aarch64 kernel.

How I got this kernel is not clear to me, I suspect I used the "rpi-update" tool once, which gave me perhaps an aarch64 version of the kernel (/boot/kernel8.img), and later it was activated by an "apt upgrade" (I had checked before, I did not update the kernel via apt prior the problem appeared).

So then I tried to get rid of the aarch64 kernel, I did a 
"WANT_64BIT=0 rpi-update"
but that did not do the trick -- I do not know if the firmware installed depends on that flag though.
Afterwards I just added "kernel=kernel7l.img" to the /boot/config.txt file, this worked.
Somehow wlan0 was working strangely and I guess an important step is to do
"sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel"
to configure everthing correctly. Since then the system is behaving as before and I can reach the octoprints again via
haproxy (portmapping to port 80) and PrintTimeGenius works properly again. 

So even though the raspi aarch64 kernel can run 32bit userspace programs fine, as well as aarch64 binaries -- the docker containers can only be 32bit (armv7l) (I also tried aarch64 octoprint image -- does not work, I guess because the docker engine is 32bit), and somehow not working perfectly fine with the aarch64 kernel, showing in this strange haproxy behaviour inside the octoprint image.

If someone has more insight on how to configure raspi to use a 32bit kernel, or a more appropriate way to "downgrade" to 32bit kernel, please correct me here -- I am no expert here and I do not want to spread "improper" information....

milutz, 2023-07-11:
Feeding off of @hemna comment that "looks like this is a problem with haproxy" and doing some googling based on @possebaer guess that haproxy was getting confused by the host having a 64bit kernel,  I found at least one other haproxy-using-package that claimed the problem went away if you upgrade haproxy

Given the dockerfile loads haproxy via apt-get (instead of building directly) the upgrading in place is kinda tricky ... that is unless the base os image itself  is upgraded, moving all the libraries forward (which can break all sorts of other things)

... but I thought I would give it a try by switching the OS image to bullseye, by changing the `ARG PYTHON_BASE_IMAG` (on the first line of the Dockerfile) to `3.8-slim-bullseye`

Happily that built ok and seemed like it might run, but the updated version of haproxy turned out to have major version update, so it (un-happily) deprecated the `reqrep` lines in the haproxy.cfg, and caused it to go into an exit-and-restart loop.

So, I tried to build again, but this time taking a guess at the way the `reqrep` lines should be updated in the 'root/etc/haproxy/haproxy.cfg' file:
replacing the "backend octoprint" `reqrep` line with `http-request replace-uri /(.*) /\1`
and the "backend webcam" `reqrep` line with `http-request replace-uri /webcam/(.*) /\1`

As a result, I have a build that seems to be working, though I'm only 10 minutes into configuring a fresh setup.  If I can get it fully setup I'll submit a PR