OctoPrint/octoprint-docker #253: Docker container with camera-streamer

isarrider, 2023-03-26:
@LongLiveCHIEF any plans to update the docker container soon
to make it possible to use camera-streamer?
I would love to test h264 hw-accel video encoding...

barrelltitor, 2024-02-28:
This is required now for Raspberry pi installs newer than February as only libcamera works due to the removal of the video memory setting from the boot file

Insomniattic, 2025-01-17:
Is there any way to get Camera Streamer installed into the Octoprint docker container? I recently decided to update my PI OS and recreate all my containers and now my camera will not connect into Octoprint. Have tried everything i can think of/find and have made no progress. Am new here so be gentle...

Paraphraser, 2025-01-17:
Have you tried:

```
$ sudo raspi-config nonint do_camera 0
$ sudo reboot
```

That enables the "legacy camera system". I run the OctoPrint container that way and I have no trouble with the camera.

Insomniattic, 2025-01-17:
> Have you tried:
> 
> ```
> $ sudo raspi-config nonint do_camera 0
> $ sudo reboot
> ```
> 
> That enables the "legacy camera system". I run the OctoPrint container that way and I have no trouble with the camera.

Yup, the MJPG-Streamer still shows no image via its console and control. It seems that it just cannot read the stream from the Host PI

Ponyboy47, 2026-08-31:
I did a little looking into this with Codex and it appears the HW-accel support is only for raspbian octoprint (ARM) right now because it relies on libcamera (a pi-specific library). These images support more than just ARM (I run them on an AMD server host machine myself). It shouldn't be too difficult to get software-based camera-streamer working for all architectures, but hardware support would likely be limited to just ARM at the moment. I only have a codex plus plan so I'm not sure I'll have the credits available to throw codex at this for a while, also since I don't have an ARM machine I wouldn't be able to fully test/validate the ARM hardware camera support.