OctoPrint/octoprint-docker #259: AttributeError: 'Babel' object has no attribute 'app'

epheterson, 2023-05-26:
Just noticed my Octoprint Docker container is no longer healthy, when I try to load the site I get:
> 503 Service Unavailable
> No server is available to handle this request.

When I check the logs I see:
```
2023-05-26 16:04:33,588 - octoprint.startup - INFO - ******************************************************************************
2023-05-26 16:04:33,590 - octoprint.startup - INFO - Starting OctoPrint 1.8.7
2023-05-26 16:04:33,590 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2023-05-26 16:04:33,590 - octoprint.startup - INFO - Reason for safe mode: problem during last startup
2023-05-26 16:04:33,590 - octoprint.startup - INFO - ******************************************************************************
2023-05-26 16:04:33,876 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2023-05-26 16:04:33,986 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2023-05-26 16:04:33,986 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2023-05-26 16:04:33,987 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2023-05-26 16:04:33,991 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2023-05-26 16:04:33,995 - octoprint.plugin.core - INFO - Loading plugins from /octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
2023-05-26 16:04:35,234 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
2023-05-26 16:04:35,289 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2023-05-26 16:04:35,308 - octoprint.plugin.core - INFO - Found 31 plugin(s) providing 17 mixin implementations, 34 hook handlers
2023-05-26 16:04:35,325 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2023-05-26 16:04:35,376 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/octoprint/plugins/bin/octoprint", line 8, in <module>
    sys.exit(main())
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/__init__.py", line 936, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/cli/server.py", line 317, in serve_command
    run_server(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/cli/server.py", line 163, in run_server
    octoprint_server.run()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/__init__.py", line 374, in run
    self._setup_i18n(app)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/__init__.py", line 1497, in _setup_i18n
    LOCALES = babel.list_translations()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 78, in fixed_list_translations
    dirs = additional_folders + [os.path.join(self.app.root_path, "translations")]
AttributeError: 'Babel' object has no attribute 'app'
Sentry is attempting to send 1 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit
```

wjcarpenter, 2023-05-27:
No solution, but "me, too". I switched back to octoprint/octoprint:1.8.7 and things are back to normal (though without the 1.9.0 upgrade, of course).

proffalken, 2023-05-30:
Quiet possibly related to https://github.com/OctoPrint/OctoPrint/issues/4744 - although 1.8.7 should have "fixed" any Babel issues AFAICT?

ahmadyusuf, 2023-08-01:
I am also having the same issue, I can see this ticket has been open since May, no fix yet?

PR #260 (2023-08-21): Fixes for 1.9 - Babel app not found
By changing the docker version from 3.8 to 3.10-slim-bullseye, it seems to fix #259.

This also updates the haproxy.cfg to the latest version requirements as there have been some changes there as well for the config. 

In the docker logs there are still issues around webcam endpoints.

2023-06-01 00:10:48,907 - octoprint.settings - WARNING - DeprecationWarning: Detected access to deprecated settings path ['webcam', 'snapshot'], returned value is derived from compatibility overlay. Please use the webcam system introduced with 1.9.0, this compatibility layer w [... cut]
closed 2023-08-21 by LongLiveCHIEF