OctoPrint/octoprint-docker #261: [BUG] validating /<path>/docker-compose.yml: (root) Additional property payload is not allowed

theschles, 2023-08-14:
**Describe the bug**
Brand-new install of Docker and Docker Compose on Raspbian Bullseye 64-bit.

```
➜ docker version
Client: Docker Engine - Community
 Version:           24.0.5
 API version:       1.43
 Go version:        go1.20.6
 Git commit:        ced0996
 Built:             Fri Jul 21 20:35:38 2023
 OS/Arch:           linux/arm64
 Context:           default

➜ docker compose version
Docker Compose version v2.20.2
```

Using `docker-compose.yml` downloaded from https://github.com/OctoPrint/octoprint-docker/tree/master/docker-compose.yml

```
➜ sudo docker compose up -d docker-compose.yml
validating /home/phil/Downloads/octoprint/docker-compose.yml: (root) Additional property payload is not allowed
```

**Container Details**

I don't have a container yet because I can't get `docker compose` to even work.

**To Reproduce**
See above

**Expected behavior**
It should install the Octoprint docker container and start it up for me.

**Screenshots**
See above

**Raspberry Pi 4 8GB (please complete the following information):**
➜ uname -a
Linux foo 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

**Smartphone (please complete the following information):**
N/A

**Additional context**
N/A

LongLiveCHIEF, 2023-08-21:
The error message you're getting suggests you have a `payload` property somewhere in your `docker-compose.yml`, and that is causing the error. 

You can easily check if your `docker-compose.yml` file is the issue by running `docker compose config`.  If you get output that looks like this:

```yaml
name: octoprint-docker
services:
  octoprint:
    image: octoprint/octoprint
    networks:
      default: null
    ports:
    - mode: ingress
      target: 80
      published: "80"
      protocol: tcp
    restart: unless-stopped
    volumes:
    - type: volume
      source: octoprint
      target: /octoprint
      volume: {}
networks:
  default:
    name: octoprint-docker_default
volumes:
  octoprint:
    name: octoprint-docker_octoprint
```

...then you're good to go.  If you get _anything_ else, then you have a syntax/tokenization problem in your compose file.

theschles, 2023-08-23:
Apologies, this is now moot.  I wiped and re-installed Raspbian -- and now it works...

closed 2023-08-23 by theschles