OctoPrint/octoprint-docker #263: upgrade to bookworm release python

idaband, 2023-10-30:
My host environment uses bookworm.  Sharing certain libraries is now having an issue when passing them into the container.

I would like the octoprint container to be based off of Bookworm

adelton, 2023-12-27:
I am able to rebuild with `--build-arg PYTHON_BASE_IMAGE=3.11-slim-bookworm` so something similar could work for you as well.

Sparticuz, 2024-01-26:
If backwards compatibility is an issue, it would be nice to publish a bookworm image in addition to bullseye.

Silvenga, 2024-04-27:
Would be good for this in any case, since Bullseye will likely end-of-life in a few months.

GlowKitty, 2025-01-27:
Seems like if you want to use this container with obico there are some issues, I haven't solved it yet but based on [this](https://www.obico.io/docs/user-guides/webcam-stream-stuck-at-1-10-fps/) documentation there are issues with images based on 3.10 bullseye with webcam streaming, and I haven't been able to get my camera to forward to obico yet using this container.