OctoPrint/octoprint-docker #282: [BUG] Upgrade from 1.10.3 to 1.11.0 503 Service Unavailable

jde6psu, 2025-04-27:
**Bug / Issue**

Upgrading to 1.11.0 fails to load in the browser. Reverted to previous container version with OctoPrint v1.10.3 restored the service.

```text
503 Service Unavailable
No server is available to handle this request.
```

Docker compose log on startup:

```text
octoprint-mini  | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
octoprint-mini  | [s6-init] ensuring user provided files have correct perms...exited 0.
octoprint-mini  | [fix-attrs.d] applying ownership & permissions fixes...
octoprint-mini  | [fix-attrs.d] done.
octoprint-mini  | [cont-init.d] executing container initialization scripts...
octoprint-mini  | [cont-init.d] 01-mjpg-streamer-config: executing... 
octoprint-mini  | [cont-init.d] 01-mjpg-streamer-config: exited 0.
octoprint-mini  | [cont-init.d] v3-dir-structure-check: executing... 
octoprint-mini  | AUTOMIGATE enabled....
octoprint-mini  | [cont-init.d] v3-dir-structure-check: exited 0.
octoprint-mini  | [cont-init.d] done.
octoprint-mini  | [services.d] starting services
octoprint-mini  | Deprecation warning: the environment variable '$MJPEG_STREAMER_INPUT' was renamed to '$MJPG_STREAMER_INPUT'
octoprint-mini  | [services.d] done.
octoprint-mini  | MJPG Streamer Version.: 2.0
octoprint-mini  |  i: Using V4L2 device.: /dev/video2
octoprint-mini  |  i: Desired Resolution: 1920 x 1080
octoprint-mini  |  i: Frames Per Second.: 30
octoprint-mini  |  i: Format............: JPEG
octoprint-mini  |  i: TV-Norm...........: DEFAULT
octoprint-mini  | 2025-04-27 10:08:45,346 - octoprint.startup - CRITICAL - Could not initialize settings manager: No module named 'class_doc'
octoprint-mini  | 2025-04-27 10:08:45,346 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
octoprint-mini  | Could not initialize settings manager: No module named 'class_doc'
octoprint-mini  | There was a fatal error starting up OctoPrint.
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
octoprint-mini  | UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
octoprint-mini  |  o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
octoprint-mini  |  o: HTTP TCP port........: 8080
octoprint-mini  |  o: HTTP Listen Address..: (null)
octoprint-mini  |  o: username:password....: disabled
octoprint-mini  |  o: commands.............: enabled
octoprint-mini  | 2025-04-27 10:08:46,317 - octoprint.startup - CRITICAL - Could not initialize settings manager: No module named 'class_doc'
octoprint-mini  | 2025-04-27 10:08:46,317 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
octoprint-mini  | Could not initialize settings manager: No module named 'class_doc'
octoprint-mini  | There was a fatal error starting up OctoPrint.
```

**Container Details**

```bash
$ docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:latest
2025-04-22T09:34:09.568Z
```

**Steps to reproduce the issue**

1. `docker compose stop; docker compose pull; docker compose up -d`
2. Go to host IP in browser, see the 503 error.

**Expected behavior**

OctoPrint starts up correctly.

**Screenshots**

[image]

Docker Container Host OS:

```bash
PRETTY_NAME="Ubuntu 24.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.2 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

**Desktop:**

 - MacOS 15.4.1 (24E263)
 - Google Chrome: Version 135.0.7049.115 (Official Build) (arm64)
 - Safari: Version 18.4 (20621.1.15.11.10)

**Workaround**

Update docker compose file to use the previous image tag.

```bash
services:
  octoprint:
    #image: octoprint/octoprint
    image: octoprint/octoprint:1.10.3
```

justindelliott, 2025-05-13:
Hi @LongLiveCHIEF  - Any recommendations on what we can do to fix this issue? Thanks!

justindelliott, 2025-05-13:
Seeing the same issue with `image: octoprint/octoprint:1.11.1` ...

```text
octoprint-mk4s  | There was a fatal error starting up OctoPrint.
octoprint-mk4s  | 2025-05-13 12:07:49,679 - octoprint.startup - CRITICAL - Could not initialize settings manager: No module named 'class_doc'
octoprint-mk4s  | 2025-05-13 12:07:49,679 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
octoprint-mk4s  | Could not initialize settings manager: No module named 'class_doc'
```

tidalvirus, 2025-06-02:
I have no idea if this is helpful, but I noticed that class_doc is in a local file on my container in `/octoprint/plugins/lib/python3.10/site-packages/octoprint/vendor/with_attrs_docs.py`

However, in the latest source, it's changed to `.class_doc` from `class_doc`.

So possibly something isn't coming down properly in the docker build for people with existing data, which is probably why it's not being noticed by more people?

@justindelliott - I haven't taken your steps yet, but I might. As you have though, what does your equivalent of that file show? `.class_doc` or `class_doc`?